tg44/OctoPrint-Prometheus-Exporter #1: Bug? Octoprint UI graph shows just Target bed temperature

ra100, 2019-12-09:
First, thank you for this plugin, I've been thinking about it long time ago, but was lacking some information and enough dedication.

And for the bug, this is what I see, since I installed this. I think it's not desired state, on the other hand, I doesn't bother me much now, when I can see the graph in grafana.

[image]

tg44, 2019-12-09:
This can be the cause of this. I push up a new version, try to force install it pls and report back if it works.

tg44, 2019-12-09:
auto close, my bad, btw its "live", can you check it?

ra100, 2019-12-09:
I'll be able to check it tomorrow, I have ~8hrs print time left :) But then, I'll report back.

tg44, 2019-12-10:
After the update it works! I still waiting for your response, but it is ok on my machine now.

ra100, 2019-12-10:
I've checked it too, looks fine, thank you.

commit df84dc81 (2019-12-09): return the original temps probably fixes #1
closed 2019-12-10 by ra100