tg44/OctoPrint-Prometheus-Exporter #2: Support when octoprint has authentication enabled

kgyrtkirk, 2019-12-11:
I've auth enabled in octoprint; so I get back FORBIDDEN when prometheus scrapes the the target 

{code}
pi@octopi:~ $ wget http://127.0.0.1:5000/plugin/prometheus_exporter/metrics
--2019-12-11 13:30:28--  http://127.0.0.1:5000/plugin/prometheus_exporter/metrics
Connecting to 127.0.0.1:5000... connected.
HTTP request sent, awaiting response... 403 FORBIDDEN
2019-12-11 13:30:28 ERROR 403: FORBIDDEN.
{code}

kgyrtkirk, 2019-12-11:
this might be a documentation-only issue; I've found a way to configure it:

```
  - job_name: 'octoprint'
    scrape_interval: 15s
    metrics_path: '/plugin/prometheus_exporter/metrics'
    params:
      apikey: ['__OCTOPRINT_APIKEY__']
    static_configs:
      - targets: ['a8:80']
```

tg44, 2019-12-12:
nice, I had no idea that; 
1. the plugin apis are protected by default
2. this easy to solve this problem :)

I will update the documentations. Thanks the contribution!

commit a0776dc5 (2019-12-12): updated documentations fixes #2
closed 2019-12-12 by tg44