tg44/OctoPrint-Prometheus-Exporter #4: Counters reset when server restarts

davidfeeney, 2020-02-04:
The counters reset whenever the server is restarted. I'm not sure if this is intended behavior, but for record keeping purposes there should be an option to retain counter values between server restarts.  Otherwise it is very hard to use these metrics practically. 

Perhaps saving the counter values to config.yaml whenever changed so they can be pulled at Octoprint startup would work. Referencing the following counter values:

started prints - as counter
failed prints - as counter
done prints - as counter
cancelled prints - as counter
timelapse count - as counter

An additional variable within config.yaml would be useful for manually resetting the counters whenever needed. This could be a binary entry that resets to default value whenever the counters are reset.

tg44, 2020-02-04:
I think this shouldn't be a problem. As I remember if you define counters in prometheus, it will handle the counterdrops, and add the new values to the old/last measured ones.

https://www.robustperception.io/how-does-a-prometheus-counter-work

It says you should use `rate()`.

tg44, 2020-02-04:
If this solves your problem, probably we can add this to the readme, or if you have a "working" dashboard you can share that too!

davidfeeney, 2020-02-04:
I'm getting a parse error when using rate(octoprint_started_prints_total):

"1:37: parse error: expected type range vector in call to function "rate", got instant vector"

tg44, 2020-02-04:
okay, you have 2 options:
 - you are interested in the "total" count in other words the sum of all events: `sum(octoprint_done_prints_total)`, if you have multiple printers probably want to somehow group this data
 - you interested in some kind of periodicity this is the `rate(octoprint_done_prints_total[30d])` this will tell you for every given time that how much you "printed" in the prev 30 days (ofc I assume that you persist data that long)

The only thing that I don't know is what happens when the data is rotated out (bcs of the retention settings), but I assume the sum and last values are saved for continuous sum and rate calculations...

I think the [basics](https://prometheus.io/docs/prometheus/latest/querying/basics/), [functions](https://prometheus.io/docs/prometheus/latest/querying/functions/) and [examples](https://prometheus.io/docs/prometheus/latest/querying/examples/) pages are worth to read, or at least I go back to them time-to-time when I want to write more complex queries.

davidfeeney, 2020-02-04:
increase(octoprint_done_prints_total[30d]) seems to be providing the behavior I am looking for. Thank you for your patience in explaining the way prometheus handles counters, this was definitely user error on my behalf so I think this can be closed now. Maybe a few of the links you provided here may be helpful to add to the readme for future reference.

closed 2020-02-05 by davidfeeney