tg44/OctoPrint-Prometheus-Exporter #6: Multiple

davidfeeney, 2020-02-05:
Multiple entries are being recorded for print progress causing an issue with the query where a new gauge is created whenever a print starts. I'm not sure if this is something that can be solved with a specific prometheus query that I'm missing or if this is an actual issue with the plugin.

Here is a copy and paste from prometheus_exporter/metrics which shows the issue:

# HELP octoprint_failed_prints_total Failed print jobs
# TYPE octoprint_failed_prints_total counter
octoprint_failed_prints_total 2.0
# TYPE octoprint_failed_prints_created gauge
octoprint_failed_prints_created 1.580942037890002e+09
# HELP octoprint_temperatures_target Targeted temperatures
# TYPE octoprint_temperatures_target gauge
octoprint_temperatures_target{identifier="B"} 0.0
octoprint_temperatures_target{identifier="T0"} 0.0
# HELP octoprint_raspberry_core_temperature Core temperature of Raspberry Pi
# TYPE octoprint_raspberry_core_temperature gauge
octoprint_raspberry_core_temperature 35.0
# HELP octoprint_infos_info Octoprint host informations
# TYPE octoprint_infos_info gauge
octoprint_infos_info{app_start="1580942042",host="SAN104",octoprint_version="1.3.12",platform="Linux"} 1.0
# HELP process_virtual_memory_bytes Virtual memory size in bytes.
# TYPE process_virtual_memory_bytes gauge
process_virtual_memory_bytes 3.33950976e+08
# HELP process_resident_memory_bytes Resident memory size in bytes.
# TYPE process_resident_memory_bytes gauge
process_resident_memory_bytes 7.7651968e+07
# HELP process_start_time_seconds Start time of the process since unix epoch in seconds.
# TYPE process_start_time_seconds gauge
process_start_time_seconds 1.58094203505e+09
# HELP process_cpu_seconds_total Total user and system CPU time spent in seconds.
# TYPE process_cpu_seconds_total counter
process_cpu_seconds_total 197.66
# HELP process_open_fds Number of open file descriptors.
# TYPE process_open_fds gauge
process_open_fds 35.0
# HELP process_max_fds Maximum number of open file descriptors.
# TYPE process_max_fds gauge
process_max_fds 1024.0
# HELP octoprint_captured_timelapses_total Timelapse captured
# TYPE octoprint_captured_timelapses_total counter
octoprint_captured_timelapses_total 0.0
# TYPE octoprint_captured_timelapses_created gauge
octoprint_captured_timelapses_created 1.580942037890099e+09
# HELP octoprint_done_prints_total Done print jobs
# TYPE octoprint_done_prints_total counter
octoprint_done_prints_total 0.0
# TYPE octoprint_done_prints_created gauge
octoprint_done_prints_created 1.580942037890034e+09
# HELP octoprint_temperatures_actual Reported temperatures
# TYPE octoprint_temperatures_actual gauge
octoprint_temperatures_actual{identifier="B"} 21.78
octoprint_temperatures_actual{identifier="T0"} 21.79
# HELP octoprint_client_num The number of connected clients
# TYPE octoprint_client_num gauge
octoprint_client_num 1.0
# HELP octoprint_slice_progress Slice progress
# TYPE octoprint_slice_progress gauge
# HELP octoprint_print_progress Print progress
# TYPE octoprint_print_progress gauge
**octoprint_print_progress{path="tweaked_hmi-plate-cr2020_9a.gcode"} 3.0
octoprint_print_progress{path="tweaked_hmi-plate-cr2020_10a.gcode"} 5.0**
# HELP octoprint_cancelled_prints_total Cancelled print jobs
# TYPE octoprint_cancelled_prints_total counter
octoprint_cancelled_prints_total 2.0
# TYPE octoprint_cancelled_prints_created gauge
octoprint_cancelled_prints_created 1.580942037890065e+09
# HELP octoprint_printer_state_info Printer connection info
# TYPE octoprint_printer_state_info gauge
octoprint_printer_state_info{state_id="OPERATIONAL",state_string="Operational"} 1.0
# HELP octoprint_started_prints_total Started print jobs
# TYPE octoprint_started_prints_total counter
octoprint_started_prints_total 2.0
# TYPE octoprint_started_prints_created gauge
octoprint_started_prints_created 1.580942037889967e+09
# HELP python_info Python platform information
# TYPE python_info gauge
python_info{implementation="CPython",major="2",minor="7",patchlevel="16",version="2.7.16"} 1.0

This is probably an issue with how I am handling the query response but I can't figure out how to filter for only the first object in the resulting array based on the documentation (the most recent entry should always be object 0 in the array). However, I'm also unsure as to why one would want to retain multiple entries for this metric since it is a gauge - hence why I'm submitting this as a potential issue.

davidfeeney, 2020-02-07:
This can be handled with a table format and calculation set to last. This will make the visualization only pull the last value added to the table which will be the most recent one.

davidfeeney, 2020-02-07:
Actually it looks like that using table format and calculation set to last will not work since new octoprint_print_progress lines are not always sorted. Going back and reprinting a job earlier in the queue resulted in a frozen progress gauge from the previous print. 

In the metrics below, I am currently printing the second line at 8% complete, however, calculation set to last will return 63% and if set to first it will return 100%. So in this scenario I see no way to query the current print progress:

`# HELP octoprint_print_progress Print progress
# TYPE octoprint_print_progress gauge
octoprint_print_progress{path="tweaked_button-binary.gco"} 63.0
octoprint_print_progress{path="hmi-plate-cr2020.gcode"} 8.0
octoprint_print_progress{path="button-binary.gco"} 100.0`

Pretty sure this is unintended behavior at this point.

tg44, 2020-02-07:
This is an interesting question that bothers me too.
The problems I wanted to solve:
 - Multiple printers
 - List of printed files
Probably I should just let the second go and delete the "path" label... Or we should unload the metric if it goes to a canceled or done state. I think that would be the "best" solution.

If you have time to give that a PR that would be nice, I will try to allocate some time to this at the end of Sunday if not.

davidfeeney, 2020-02-08:
I am actually using the path label - so it would be beneficial to keep that. As you suggested, I agree the best path will likely be to unload the metric if it goes to a canceled or complete state. That would solve the problem and allow the active path to still be monitored.

tg44, 2020-02-11:
Sorry, I don't really have time to do things other than work. I try to get some time to fix this in the next 2 weeks, but I can't agree anything right now...

tg44, 2020-02-13:
I just merged a pr, I think this will fix this issue. Can you confirm it?

rboutinaud, 2020-02-15:
Canceled prints are still being recorded after the 0.1.2 update 
```
# TYPE octoprint_print_progress gauge
octoprint_print_progress{path="prusa_screen_cover_0.1mm_PETG_MK3S_46m.gcode"} 0.0
octoprint_print_progress{path="prusa_screen_cover_0.1mm_PETG_MK3S_47m.gcode"} 2.0
```

tg44, 2020-02-16:
probably now

rboutinaud, 2020-02-16:
Working now, cancelled prints are no longer showing

closed 2020-02-16 by tg44