tg44/OctoPrint-Prometheus-Exporter #7: Singlestat usage with state_id

davidfeeney, 2020-02-08:
The `state_id` field from metric `octoprint_printer_state_info` is dropped during octoprint server restart. This causes a problem with Stinglestat panel visualization of the query response in Grafana. 

The issue of missing field seems to cause Grafana to revert the singlestat field to returning another field (in my case `__name__`) at every octoprint restart since the `state_id` field cannot be found. I would then need to manually change the field back to `state_id` once it becomes available again (once reconnected to the printer).

State_id (and therefore state_string also) should be initialized at plugin startup to avoid this issue. The actual name of the startup status/value isn't too important, just that something is assigned to `state_id` at plugin startup so the field does not disappear and cause a visualization problem.

tg44, 2020-02-11:
Yapp, I found that too... And also I think the correct fix would be sending a stat not as a property but rather as an id/number... there is a panel type called discrete, and that would be super nice if I would just send a number as "state".
BTW I used this as a quickfix in graphana, no idea if it fixes your problem or not:
`octoprint_printer_state_info{state_id!=""}`

davidfeeney, 2020-02-15:
Your quickfix worked perfectly for me, thanks! Afterwards I set the null value in the visualization value mappings to "OFFLINE" and the panel transitions smoothly now.

closed 2020-02-15 by davidfeeney

jneilliii, 2020-07-31:
> Afterwards I set the null value in the visualization value mappings to "OFFLINE" and the panel transitions smoothly now.
could you expand on this please @davidfeeney. I'm totally new to Grafana and would like to implement this fix in my dashboard.