tg44/OctoPrint-Prometheus-Exporter #10: Endpoint Forbidden

vaddi, 2020-04-01:
Hi there,

im unable to get the metrics into my prometheus. 
After installing the Plugin, i've opened the Endpoint in another tab and saw all metrics. So i configure the new Target into my Prometheus System which runs on another host. But here i only get  the Error:
server returned HTTP status 403 FORBIDDEN

I've then tried to get the Metrics manual via curl to ensure there is no missconfig on Prometheus:
curl http://octopi/plugin/prometheus_exporter/metrics
Butt also get only "Forbidden". 

I set an API Key for the Read-Only user and use them as Token:
curl -H "X-Api-Key: XXXXXXXXX" http://octopi/plugin/prometheus_exporter/metrics
# HELP octoprint_z_travel_print Z axis travel in this print
# TYPE octoprint_z_travel_print gauge
octoprint_z_travel_print 0.0
# HELP octoprint_print_fan_speed Fan speed
# TYPE octoprint_print_fan_speed gauge
octoprint_print_fan_speed 0.0
...

It seems that i need to pass a api-key to my prometheus Target, but i only find the "base auth" possibility. On the OctoPrint i've also tried to allow access to the Endpoint without any authentication, but didn't find anything ether.

vaddi, 2020-04-02:
I read this Repo README.md and found the missing config param:

```
    params:
      apikey: ['__OCTOPRINT_APIKEY__']
```
This solves my problem.

vaddi, 2020-04-02:
Maybe another one came here with the same Problem. 
I've close this issue

closed 2020-04-02 by vaddi

tg44, 2020-04-02:
I love when a readme can solve problems :D