tg44/OctoPrint-Prometheus-Exporter #13: Duplicated timeseries in CollectorRegistry: set(['octoprint_temperatures_actual'])

jangrewe, 2020-04-30:
Hi, i love your plugin, but on one of my machines this showed up and seems to kill OctoPrint (it happened when installing another plugin):
```
2020-04-30 13:43:41,127 - octoprint.plugin.core - ERROR - Error loading plugin prometheus_exporter
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 967, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 71, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prometheus_exporter/__init__.py", line 18, in <module>
    octoprint.plugin.EventHandlerPlugin):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_prometheus_exporter/__init__.py", line 42, in PrometheusExporterPlugin
    temps_actual = Gauge('octoprint_temperatures_actual', 'Reported temperatures', ['identifier'])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/prometheus_client/metrics.py", line 324, in __init__
    labelvalues=labelvalues,
  File "/home/pi/oprint/local/lib/python2.7/site-packages/prometheus_client/metrics.py", line 107, in __init__
    registry.register(self)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/prometheus_client/registry.py", line 29, in register
    duplicates))
ValueError: Duplicated timeseries in CollectorRegistry: set(['octoprint_temperatures_actual'])
```

tg44, 2020-05-03:
You restarted the octopi instance and it solved?

I think this is probably a double inicialization error somewhere. I'm not a pro in python, nor in octoprint, nor the python prometheus lib, but I think the following happened:
 - you started the octoprint 
    - it started the prometheus plugin which initialized a time series
 - you installed a new plugin
 - octoprint reloaded all the plugins
    - it started the prom plugin again, which tried to initialize the time series again

I think to resolve this problem we should deregister the timeseries when the plugin reloads, or move all of our metrics inicialization to a "check if exist and register if not" block...

I think this is not a high probability or showstopper issue. I tag it with a "good first contrib" tag. I will not have time to do it in the seeable future :(

tg44, 2020-08-16:
Can you test this, and report back if installing new plugins besides this working for you too without the reported errors? (after updating to 1.5)

commit 8776b472 (2020-08-16): moved to local registry usage (fixes plugin reload), fixes #13

moved to local registry usage (fixes plugin reload), fixes #13
closed 2020-08-16 by tg44