tg44/OctoPrint-Prometheus-Exporter #15: Enhancement - PrintTime Genius Time Remaining

tideline3d, 2020-07-31:
Would it be possible to publish the time remaining from the Print Time Genius plugin through Prometheus?  Would love to have that on my Grafana dashboard.

tg44, 2020-08-01:
Based on the docs, not really possible, but I asked it here; https://community.octoprint.org/t/is-remaining-time-available-as-an-event-hook/22579

tideline3d, 2020-08-13:
No worries if its not possible.  Can probably get that data into Grafana directly through the API.

tg44, 2020-08-13:
Based on the info I get from `jneilliii` it is possible, I just didn't had time to implement and try it out yet.

tg44, 2020-08-16:
Can you test this, and report back if it is working for you too?

closed 2020-08-16 by tg44

tideline3d, 2020-08-16:
Installed now, will follow it with my prints today.  Thanks!

commit 62f1172b (2020-08-16): expose elapsed time, estimation and left estimation, fixes #15