tg44/OctoPrint-Prometheus-Exporter #16: Temperatures report last known after error

tideline3d, 2020-08-13:
I had a thermal runaway error overnight and the data from this plugin was incredibly useful to see that something indeed went wrong and it wasn't just a false alarm.   Thanks again for this plugin.

I noticed though after my printer went into error state and I powered it off, the exporter is still reporting the last known temperature to Prometheus.  I'd expect this to either be null, some error value or just zero.

Not a common problem or easily reproducible but thought I'd share my experience.

[image]

tideline3d, 2020-08-15:
Looking into this some more, it seems any time the printer disconnects the last temp value remains.  (I have mine turn off automatically on idle with the TPLink plugin).  I assume this is a feature of the time-series data structure.  It'd be nice if on printer disconnect for any reason, the prometheus-exporter sent out a 0 for temp.

tg44, 2020-08-15:
Yapp, its on my list right after the estimated time measure :D But ofc if someone want to work on it its free to take and actually a good first issue.

We need to detect printer status changes, and if the status is changed to DC we should stop exposing temp data (making a hole in the graph instead of falsely pull it to 0) if its possible.

tideline3d, 2020-08-15:
Good point on 0 vs empty. 0 is technically a valid temperature.

tg44, 2020-08-16:
Can you test this, and report back if it is working for you too?

closed 2020-08-16 by tg44

tideline3d, 2020-08-16:
Installed today, will follow up.

tideline3d, 2020-08-16:
Looks to be working well.  Cut off my printer today in anger at a first layer and it was caught on the graphs 👍 

[image]

commit a28bd8b8 (2020-08-16): deregister temperature metrics when the printer goes offline fixes #16