tg44/OctoPrint-Prometheus-Exporter #17: Prometheus can't import data

sbaloo, 2020-09-04:
Hi surely i forget something. Prometheus is configure, the dashboard in grafana also but in the prometheus log i have:
level=warn ts=2020-09-04T09:19:47.522855635Z caller=scrape.go:836 component="scrape manager" scrape_pool=octoprint target="http://xxx.xxx.xxx:5001/plugin/prometheus_exporter/metrics?apikey=xxxxxxx" msg="append failed" err="invalid metric type \"info\""

Can you help me?

sbaloo, 2020-09-04:
Just to give more clues
# HELP octoprint_temperatures_actual Reported temperatures
# TYPE octoprint_temperatures_actual gauge
# HELP octoprint_temperatures_target Targeted temperatures
# TYPE octoprint_temperatures_target gauge
# HELP octoprint_raspberry_core_temperature Core temperature of Raspberry Pi
# TYPE octoprint_raspberry_core_temperature gauge
octoprint_raspberry_core_temperature 0.0
# HELP octoprint_infos_info Octoprint host informations
# TYPE octoprint_infos_info gauge
octoprint_infos_info{app_start="1598946064",host="4b5470b0fc88",octoprint_version="1.4.2",platform="Linux"} 1.0
# HELP octoprint_client_num The number of connected clients
# TYPE octoprint_client_num gauge
octoprint_client_num 0.0
# HELP octoprint_printer_state_info Printer connection info
# TYPE octoprint_printer_state_info gauge
octoprint_printer_state_info{state_id="OFFLINE",state_string="Offline"} 1.0
# HELP octoprint_started_prints_total Started print jobs
# TYPE octoprint_started_prints_total counter
octoprint_started_prints_total 4.0
# HELP octoprint_started_prints_created Started print jobs
# TYPE octoprint_started_prints_created gauge
octoprint_started_prints_created 1.598946059415033e+09
# HELP octoprint_failed_prints_total Failed print jobs
# TYPE octoprint_failed_prints_total counter
octoprint_failed_prints_total 1.0
# HELP octoprint_failed_prints_created Failed print jobs
# TYPE octoprint_failed_prints_created gauge
octoprint_failed_prints_created 1.5989460594150527e+09
# HELP octoprint_done_prints_total Done print jobs
# TYPE octoprint_done_prints_total counter
octoprint_done_prints_total 3.0
# HELP octoprint_done_prints_created Done print jobs
# TYPE octoprint_done_prints_created gauge
octoprint_done_prints_created 1.59894605941507e+09
# HELP octoprint_cancelled_prints_total Cancelled print jobs
# TYPE octoprint_cancelled_prints_total counter
octoprint_cancelled_prints_total 0.0
# HELP octoprint_cancelled_prints_created Cancelled print jobs
# TYPE octoprint_cancelled_prints_created gauge
octoprint_cancelled_prints_created 1.5989460594150877e+09
# HELP octoprint_captured_timelapses_total Timelapse captured
# TYPE octoprint_captured_timelapses_total counter
octoprint_captured_timelapses_total 0.0
# HELP octoprint_captured_timelapses_created Timelapse captured
# TYPE octoprint_captured_timelapses_created gauge
octoprint_captured_timelapses_created 1.598946059415105e+09
# HELP octoprint_print_progress Print progress
# TYPE octoprint_print_progress gauge
# HELP octoprint_print_time_elapsed Print time elapsed
# TYPE octoprint_print_time_elapsed gauge
# HELP octoprint_print_time_est Print time estimate
# TYPE octoprint_print_time_est gauge
# HELP octoprint_print_time_left_estimate Print time left estimate
# TYPE octoprint_print_time_left_estimate gauge
# HELP octoprint_slice_progress Slice progress
# TYPE octoprint_slice_progress gauge
# HELP octoprint_printing_time_total Printing time total
# TYPE octoprint_printing_time_total counter
octoprint_printing_time_total 34536.834331035614
# HELP octoprint_printing_time_created Printing time total
# TYPE octoprint_printing_time_created gauge
octoprint_printing_time_created 1.598946059415216e+09
# HELP octoprint_extrusion_total Filament extruded total
# TYPE octoprint_extrusion_total counter
octoprint_extrusion_total 8.512277625402308e+08
# HELP octoprint_extrusion_created Filament extruded total
# TYPE octoprint_extrusion_created gauge
octoprint_extrusion_created 1.5989460594152384e+09
# HELP octoprint_extrusion_print Filament extruded this print
# TYPE octoprint_extrusion_print gauge
octoprint_extrusion_print 0.0
# HELP octoprint_x_travel_total X axis travel total
# TYPE octoprint_x_travel_total counter
octoprint_x_travel_total 594667.161999967
# HELP octoprint_x_travel_created X axis travel total
# TYPE octoprint_x_travel_created gauge
octoprint_x_travel_created 1.598946059415411e+09
# HELP octoprint_x_travel_print X axis travel in this print
# TYPE octoprint_x_travel_print gauge
octoprint_x_travel_print 0.0
# HELP octoprint_y_travel_total Y axis travel total
# TYPE octoprint_y_travel_total counter
octoprint_y_travel_total 557402.9559999047
# HELP octoprint_y_travel_created Y axis travel total
# TYPE octoprint_y_travel_created gauge
octoprint_y_travel_created 1.5989460594154484e+09
# HELP octoprint_y_travel_print Y axis travel in this print
# TYPE octoprint_y_travel_print gauge
octoprint_y_travel_print 0.0
# HELP octoprint_z_travel_total Z axis travel total
# TYPE octoprint_z_travel_total counter
octoprint_z_travel_total 591.3199999999997
# HELP octoprint_z_travel_created Z axis travel total
# TYPE octoprint_z_travel_created gauge
octoprint_z_travel_created 1.5989460594154873e+09
# HELP octoprint_z_travel_print Z axis travel in this print
# TYPE octoprint_z_travel_print gauge
octoprint_z_travel_print 0.0
# HELP octoprint_print_fan_speed Fan speed
# TYPE octoprint_print_fan_speed gauge
octoprint_print_fan_speed 0.0

tg44, 2020-09-06:
If you ctrl+f for `info` in this page, it only appears to places where it shouldd be, so I have no idea at this point. If you can repro this, based on the error we should have a
```
# TYPE something_something info
```
line. I use a lib to generate this page, so I have really no idea what is the problem from the given information.

But you should try to paste the output link (from the log) to a private browser window. It is possible that the page you saw is not the same as the prometheus instance see. This is my best idea right now...

tg44, 2020-09-28:
Is this still a thing? Or can I close it?

tg44, 2020-10-04:
I close this due to missing informations and inactivity. If it is still happening pls let me know!

closed 2020-10-04 by tg44