tg44/OctoPrint-Prometheus-Exporter #18: Error processing gcodes

mafredri, 2020-09-24:
Hi,

I'm not sure when I've started seeing this in the logs, but I recently upgraded to Python 3 and believe that was when. The following error is being spammed into the log for every line of gcode issues during the print.

```
2020-09-24 20:00:38,059 - octoprint.util.comm - ERROR - Error while processing hook prometheus_exporter for phase sent and command G1 X102.432 Y140.005 E0.01962:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 3435, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_prometheus_exporter/__init__.py", line 267, in gcodephase_hook
    self.print_time_est.labels(self.print_progress_label).set(data['job']['estimatedPrintTime'])
  File "/home/pi/oprint/lib/python3.7/site-packages/prometheus_client/metrics.py", line 359, in set
    self._value.set(float(value))
TypeError: float() argument must be a string or a number, not 'NoneType'
```

Since it seems to be erroring on the print time estimate, I'm guessing it's some bad interplay with PrintTimeGenious? I'll need to take a look if it's actually working properly, but it may also be worth handling it in the code instead of throwing an exception?

tg44, 2020-09-25:
Interesting... I tried it out with PTG and py3 as I remember. Probably somewhere something is not providing a valid value that I expect... I'm not close to computers in the next couple of days, so feel free to dig in it if you have time.

ra100, 2020-09-28:
I just wanted to report the same. I upgraded several days ago, and now noticed same error in octoprint.log, when checking for something else.
I'm getting it right now on:
```
2020-09-28 16:15:56,994 - octoprint.util.comm - ERROR - Error while processing hook prometheus_exporter for phase sent and command M27:
```
I'm printing from SD right now.

ra100, 2020-10-04:
I'd like to look at this issue.
Seems as a side effect, it's causing print stutter too (on Pi Zero).

tg44, 2020-10-04:
I think it is fixed with the latest version. Pls reopen this if not.

closed 2020-10-04 by tg44