tg44/OctoPrint-Prometheus-Exporter #20: Prometheus alerting that service is down even though data can be read

JohSchillerEmnify, 2020-12-17:
My setup is as follows:

## octoprint
```
version: '2.4'

services:
  octoprint:
    image: octoprint/octoprint
    restart: unless-stopped
    ports:
      - 5000:80
    devices:
      - /dev/ttyACM0:/dev/ttyACM0
      - /dev/video0:/dev/video0
    volumes:
      - octoprint:/octoprint
      - ./docker/octoprint_data:/root/.octoprint/
    # uncomment the lines below to ensure camera streaming is enabled when
    # you add a video device
    environment:
      - ENABLE_MJPG_STREAMER=true
```
## Prometheus scrape config

``` 
  - job_name: 'octoprint'
    scrape_interval: 5s
    metrics_path: '/plugin/prometheus_exporter/metrics'
    params:
      apikey: ['__myapikey__']
    static_configs:
      - targets: ['octoprint:80']
```
even though I have tried  it without api key, and the scrape target (htttp://ipofoctoprintserver:portofoctoprintserver/plugin/prometheus_exporter/metrics)  is reachable and seems to display correct data. 

I have also tried to change the `static_configs: target`:

- `octoprint_octoprint_1:5000` (the container name I get from running docker ps and the port I configured)
- `127.0.0.1:5000` (reachable and displaying correct data)
- `octoprint:5000` 
 
all this still results in this:
[image]

unfortunately i'm not sure how to look for issues, I'm happy to provide any necessary data needed and help fixing this if I find a solution.

tg44, 2020-12-18:
I think it is not software but rather some networking issue.

I would recommend you do the bridging thing first, but I think we should go deeper...

You can check things with [netcat in chat mode](https://www.varonis.com/blog/netcat-commands/#chat). You don't need to actively run `nc` in the target containers, you can start other containers just try to figure out what works and what is not.

I use host-bindings and domains without firewalls (it is in a safe nat anyway). But if they are in the same compose file they should see each other by "name". Also if they are not in the same compose file, but you want to host-bind the octoprint instance, you probably can reach it with [one of these methods](https://stackoverflow.com/questions/48546124/what-is-linux-equivalent-of-host-docker-internal). I'm sure that "pinging" will be false-positive in a lot of cases.

JohSchillerEmnify, 2020-12-18:
Thank you for taking the time to reply, I'll continue investigating based on your advice. They currently aren't in the same file, I tried using the virtual network interface that docker created with no success so far. 

When I find a Solution I'll definitely Update here.

closed 2020-12-18 by JohSchillerEmnify