tg44/OctoPrint-Prometheus-Exporter #21: Error while processing hook prometheus_exporter for phase sent

rtoma, 2020-12-28:
Hi, first of thanks for creating this plugin. I was browsing my octoprint logs in `~/.octoprint/logs/octoprint.log` and while printing I noticed a constant stream of messages like:

```
2020-12-28 13:28:24,436 - octoprint.util.comm - ERROR - Error while processing hook prometheus_exporter for phase sent and command G0 F9000 X86.742 Y83.612:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4490, in _process_command_phase
    tags=tags,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_prometheus_exporter/__init__.py", line 267, in gcodephase_hook
    self.print_time_est.labels(self.print_progress_label).set(data['job']['estimatedPrintTime'])
  File "/home/pi/oprint/lib/python3.7/site-packages/prometheus_client/metrics.py", line 359, in set
    self._value.set(float(value))
TypeError: float() argument must be a string or a number, not 'NoneType'
2020-12-28 13:28:24,480 - octoprint.util.comm - ERROR - Error while processing hook prometheus_exporter for phase sent and command G1 F2355.7 X86.791 Y83.427 E2644.49924:
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/comm.py", line 4490, in _process_command_phase
    tags=tags,
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_prometheus_exporter/__init__.py", line 267, in gcodephase_hook
    self.print_time_est.labels(self.print_progress_label).set(data['job']['estimatedPrintTime'])
  File "/home/pi/oprint/lib/python3.7/site-packages/prometheus_client/metrics.py", line 359, in set
    self._value.set(float(value))
TypeError: float() argument must be a string or a number, not 'NoneType'
```

This events are slowly filling and wearing my RPI's SD card. Wondering what I can do to stop this flood?

Versions:
- OctoPrint 1.5.2
- OctoPi 0.17.0 on RPI 3 Model B Rev 1.2
- Prometheus Exporter Plugin (0.1.5) = /home/pi/oprint/lib/python3.7/site-packages/octoprint_prometheus_exporter

If you need more info, please let me know.

tg44, 2020-12-28:
We probably need one more nullchecks at `__init__.py", line 267`... The strange thing that we actually [have it](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/__init__.py#L266).

But; our [plugin version](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/setup.py#L17https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/setup.py#L17) is actually 0.1.6. I think it is fixed in the [last version](https://github.com/tg44/OctoPrint-Prometheus-Exporter/commit/461d2f6d1d16145858ff1769f05117625c63e407). Can you confirm it?

rtoma, 2020-12-28:
Thanks for the quick answer. It seems the 0.16 release is not yet available. I checked for plugin updates in octoprint, but no luck. I will close knowing a fix is forthcoming.

closed 2020-12-28 by rtoma

tg44, 2020-12-29:
Actually, I learned that I need to release the new versions on github, and just updating the "version" string is not enough... It should appear in 24h, or you can force a refresh in the advanced menu.