tg44/OctoPrint-Prometheus-Exporter #25: octoprint_extrusion_print value?

robinsondan87, 2021-03-07:
This is not an issue as such but there does not seem to be any documentation on what value is displayed for octoprint_extrusion_print

EG
octoprint_extrusion_print 360904.07021000056

what is this value in?

tg44, 2021-03-07:
https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/__init__.py#L207

and

https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/gcodeparser.py#L106

Based on these, this is the extruded filament in millimeters.

philosowaffle, 2021-04-08:
Are you sure it isn't measured in something smaller than millimeters?

That means @robinsondan87 single print used 360meters of filament, if I'm understanding the counter correctly.

My reasoning:
1. [Google is telling a spool of filament has ~330m](https://www.google.com/search?q=how+many+meters+of+filament+in+1kg&oq=how+many+meters+of+fila&aqs=chrome.0.0j69i57j0j0i22i30j0i390.3795j0j7&sourceid=chrome&ie=UTF-8) of material
2. My current octoprint_extrusion_print (about 4hrs into printing) = 87145586.30048087
3. That means this one print is 87km of filament
4. But if the metrics is in micrometers then it turns into a more reasonable (yet still oddly high) 90m
5. And nanometers would yield 87mm

Its been a long day, so maybe I'm doing something stupid with the numbers here, just thought I'd see if you can sanity check me on this?

robinsondan87, 2021-04-09:
> Are you sure it isn't measured in something smaller than millimeters?
>
> That means @robinsondan87 single print used 360meters of filament, if I'm understanding the counter correctly.
>
> My reasoning: [...]

This was my thought exactly in regards to the numbers they just didn’t add up. Micrometers makes more sense and knowing this it’s now easy to convert into usable numbers now on my custom dash.

I’ll confirm this later on and may be worth adding it to the documentation? 

[image]

tg44, 2021-04-09:
I reopened this until we clarify it :D

So my second link to the code says when we get an E 12 we add 12 to the extruded value.
Based on the [Wikipedia](https://reprap.org/wiki/G-code#G0_.26_G1:_Move) it means we move/extrude 12 units. The unit here is either mm or inch based on the gcode/printer default/settings.

This code is not really straightforward :D [extrusion_counter](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/52ee0cb549c60786ba4a93c821f38d4506a53191/octoprint_prometheus_exporter/gcodeparser.py#L106) is the extruder positions sum (So E100 and E200 will result an extrusion_counter of 300.) [This](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/__init__.py#L229) set is setting this value which is kinda meaningless... This seems to be a bug, and that should be `self.parser.e`. But [this](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/__init__.py#L229) seems to be ok.

tg44, 2021-04-09:
I pushed a new plugin version. Can sb test this out?

philosowaffle, 2021-04-09:
Just installed the update. Have a print lined up for today, will monitor and report back.

That Wikipedia link is very helpful.

robinsondan87, 2021-04-09:
> I pushed a new plugin version. Can sb test this out?

I’ll give this a go later also when I’m ready to start a new print. Thanks once again for your time and effort looking at something quite trivial.

philosowaffle, 2021-04-10:
First off, want to echo @robinsondan87 , thank you for taking the time to look at this as it really is not a big deal.

Here is what I am still observing on my latest prints.  Attached screenshots as well, just in case I am misinterpreting any of the raw data.

**Plugin Version: 0.1.7**
**octoprint_extrusion_total:** 187786516.6330472 mm => 187.78651663304719932 kilometers
**octoprint_extrusion_print:** 1699017.9306399974 mm => 1.6990179306399972869 kilometers

For reference, according to the octoprint dashboard:
**Current Elapsed Print Time:** ~49min
**Extruded Filament:** 1.68 meters

If the octoprint dashboard is to be trusted, then that would imply that the metrics are actually measured in micrometers instead of millimeters.

[image]
[image]
[image]

tg44, 2021-04-15:
Do we have more data on this? At least the nembers started to match :)

philosowaffle, 2021-04-15:
That data has remained consistent for me over the next couple prints... and then my bowden tube broke so I'm waiting on a replacement part to arrive.

Seems to me, if the octoprint dashboard is measuring correctly, then we can safely say the Prometheus metric is measuring in micrometers.

tg44, 2021-05-27:
I think at this point we can close this.

aaaidan, 2021-10-29:
Okay great, I'll give it a go when I find time. Any tips/links for setting up a dev environment?

It's a relief that we agree inches are the morally superior unit. 🇺🇸 🎆 🌭

aaaidan, 2021-10-29:
I see you have good instructions and a docker environment for dev, thx!

PR #40 (2024-02-01): Bugfixes for a number of things
This PR contains a bunch of bugfixes thrown together, let me know if you'd prefer separate PRs for these

- Development stuff
  - Add gcodeparser.py entry to editorconfig
  - Set max_line_length in editorconfig
  - Add .flake8 file to suppress warnings from pylsp

- Print time metric issue
  - Fix wrong attribute being used for print time metrics resulting in missing `octoprint_print_time_*` metrics

- setup.py issue
  - Bump plugin version to 0.2.2 (fixes #38)

- G-code parsing
  - Implement coordinate mode switch detection (`G90`/`G91`/`M82`/`M83`) in gcodepar [... cut]
closed 2024-02-01 by tg44