tg44/OctoPrint-Prometheus-Exporter #31: Having trouble getting the permssions to work properly on a simple curl

jgrossmac, 2022-08-03:
```
/ # curl 192.168.10.100/plugin/prometheus_exporter/metrics
{"error":{"dangerous":false,"default_groups":["users"],"description":"Allow access to Prometheus metrics. Includes the Status permission.","key":"PLUGIN_PROMETHEUS_EXPORTER_METRICS","name":"Prometheus Exporter Plugin: Metrics access","needs":{"role":["plugin_prometheus_exporter_metrics","status"]},"plugin":"prometheus_exporter"}}
```

[image]

loelkes, 2022-08-03:
@jgrossmac I can confirm that. I will look into it (I wrote that piece) and for some magical reasons that I cannot explain right now this did work in my tests.

loelkes, 2022-10-22:
It seems this issue is related to OctoPrint. I opened an issue octoprint/octoprint#4670

loelkes, 2022-10-23:
So the solution with curl would be

1. `curl -X POST -b /tmp/octoprint-cookie.txt -c /tmp/octoprint-cookie.txt localhost:5000/api/login?passive=true`
2. `curl -b /tmp/octoprint-cookie.txt -c /tmp/octoprint-cookie.txt localhost:5000/plugin/prometheus_exporter/metrics`

But AFAIK this does not work with prometheus. In this context it's a _won't fix_ and and API key is always required. Setting individual permissions per user (access/no access) for metrics works fine.

tg44, 2022-10-23:
This is really interesting. I don't see a reason why we need a cookie for unauthenticated/public requests...

loelkes, 2022-11-02:
This is a confirmed bug in OctoPrint and will be fixed in 1.9.0.

loelkes, 2024-01-01:
@jgrossmac This was fixed, I can't reproduce it with OctoPrint 1.9.1.

Please close the issue or comment if this problem persists.

tg44, 2024-01-02:
Thanks for the heads up, I'm closing this!

closed 2024-01-02 by tg44