tg44/OctoPrint-Prometheus-Exporter #35: Question: in the head of metrics.py you are collecting RPi system core temp but it's never used.

m-arx, 2023-01-13:
Did you plan on exporting also the core temp as a metric? I was looking for this feature in particular, as I want to monitor my octopi - Pi Hardware a little closer as well...

tg44, 2023-01-16:
As I remember we [have this already](https://github.com/tg44/OctoPrint-Prometheus-Exporter/blob/master/octoprint_prometheus_exporter/metrics.py#L20).

m-arx, 2023-01-16:
yes, but it's not in the list of metrics. I have added the vgencmd to sudoers.d config as well.

tg44, 2023-01-18:
Can you check the logs if you have a `Raspberry core temperature is not supported on this system` message in it?

m-arx, 2023-01-20:
even after adding DEBUG for octoprint.plugins.prometheus_exporter, I have no log at all :/ 

I am running Octoprint 1.8.6 (in an octopi installation)

loelkes, 2024-01-01:
The message should appear in INFO logs.

Can you run
```
sudo /usr/bin/vcgencmd measure_temp
```
and paste the exact output here?

This is the exact command that the plugin executes.

m-arx, 2024-01-01:
meanwhile I have reinstalled octopi and now I have "not supported on this system" in octoprint.log.

running the command above as pi, return is:

`temp=37.0'C`

loelkes, 2024-01-01:
Do you see `Failed to execute "sudo /usr/bin/vcgencmd"` somewhere in the logs?

m-arx, 2024-01-01:
I found it - I was missing a sudoers entry to allow execution of /usr/bin/vcgencmd without pwd.

m-arx, 2024-01-01:
created /etc/sudoers.d/vcgencmd file with content

`pi ALL=NOPASSWD: /usr/bin/vcgencmd *`

closed 2024-01-01 by m-arx

loelkes, 2024-01-01:
I've included a documentation for this in my commits for an upcoming PR. See https://github.com/loelkes/OctoPrint-Prometheus-Exporter/commit/15f562e0603c60a13f82bdee1106f5649b14b9e0